RobTillaart/SHT2x #7: Help calling begin() with separate I2C interfaces

chiebert, 2022-03-18:
Help request (relatively new to I2C and C++): I see that there's an option to Begin using a specific I2C interface, but I'm lost trying to pass in a TwoWire to the Begin call. I'm trying to use two SHT21s with an ESP32, and I've already set up two TwoWire instances and two sensors via:

```
TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);
SHT2x internal;
SHT2x external;
```

In my setup() I've initiated the two TwoWire object (I think) with:

```
I2Cone.begin(SDA_1, SCL_1, 100000);
I2Ctwo.begin(SDA_2, SCL_2, 100000);
```

And I've tried initiating the two sensors with:

```
internal.begin(I2Cone);
external.begin(I2Ctwo);
```

But when compiling in PlatformIO I get errors about 'no matching function for call to 'SHT2x::begin(TwoWire&)'. Any hints as to what I'm missing?

RobTillaart, 2022-03-18:
Thanks for the issue, will answer asap but might be a few days.

RobTillaart, 2022-03-18:
You need an & I guess.
Can you test this sketch - it compiles without error messages.

If it works I will add it to the examples

```cpp
//
//    FILE: SHT2x_demo_ESP32_2.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo usage 2 I2C ports + SHT2X
//     URL: https://github.com/RobTillaart/SHT2x

#include "Wire.h"
#include "SHT2x.h"

TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);
SHT2x internal;
SHT2x external;

uint32_t start;
uint32_t stop;

SHT2x sht;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("SHT2x_LIB_VERSION: \t");
  Serial.println(SHT2x_LIB_VERSION);

  internal.begin(&I2Cone);
  external.begin(&I2Ctwo);

  uint8_t stat = internal.getStatus();
  Serial.print(stat, HEX);
  Serial.println();
  stat = external.getStatus();
  Serial.print(stat, HEX);
  Serial.println();
  Serial.println();
}

void loop()
{
  start = micros();
  internal.read();
  external.read();
  stop = micros();

  Serial.print("\t");
  Serial.print(stop - start);
  Serial.print("\t");
  Serial.print(internal.getTemperature(), 1);
  Serial.print("\t");
  Serial.print(internal.getHumidity(), 1);
  Serial.print("\t");
  Serial.print(external.getTemperature(), 1);
  Serial.print("\t");
  Serial.println(external.getHumidity(), 1);
  delay(1000);
}

// -- END OF FILE --
```

chiebert, 2022-03-18:
Thanks for the very quick help - and with one change that worked. (I was sure I had tried using the &, but it was getting late and I think I was going in circles and confusing myself.)

In your version, you didn't pass new SDA and SCL pins to the I2Ctwo interface and 'begin' it separately, so only the sensor using I2Cone (which I guess auto-began with the ESP32 default pins of 21 and 22) worked. The second sensor reported a meaningless constant result (-40 for temp and -6 for humidity), I assume because it was trying to use the second bus which was itself trying to use the same default SDA/SCL pins as the first one.

The following worked perfectly for me:

```
//
//    FILE: SHT2x_demo_ESP32_2.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo usage 2 I2C ports + SHT2X
//     URL: https://github.com/RobTillaart/SHT2x

// #include <Arduino.h> // for PlatformIO
#include "Wire.h"
#include "SHT2x.h"

#define SDA_1 21
#define SCL_1 22
#define SDA_2 33
#define SCL_2 32

TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);
SHT2x internal;
SHT2x external;

uint32_t start;
uint32_t stop;

SHT2x sht;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("SHT2x_LIB_VERSION: \t");
  Serial.println(SHT2x_LIB_VERSION);

  I2Cone.begin(SDA_1, SCL_1);
  I2Ctwo.begin(SDA_2, SCL_2);

  internal.begin(&I2Cone);
  external.begin(&I2Ctwo);

  uint8_t stat = internal.getStatus();
  Serial.print(stat, HEX);
  Serial.println();
  stat = external.getStatus();
  Serial.print(stat, HEX);
  Serial.println();
  Serial.println();
}

void loop()
{
  start = micros();
  internal.read();
  external.read();
  stop = micros();

  Serial.print("\t");
  Serial.print(stop - start);
  Serial.print("\t");
  Serial.print(internal.getTemperature(), 1);
  Serial.print("\t");
  Serial.print(internal.getHumidity(), 1);
  Serial.print("\t");
  Serial.print(external.getTemperature(), 1);
  Serial.print("\t");
  Serial.println(external.getHumidity(), 1);
  delay(1000);
}
 
// -- END OF FILE --
```

And using this principle, my original project is back on track. Thanks again!

chiebert, 2022-03-18:
Although, I do think the extra `SHT2x sht` up at the beginning just before `setup()` is superfluous, and came along for the ride from one of your other examples... Forgot to mention that and remove it from my test...

closed 2022-03-18 by chiebert