RobTillaart/SHT2x #9: Wrong values reported due to incorrect I2C pin configuration on current ESP32 Arduino API

jannvck, 2022-06-20:
In the current ESP32 Arduino API the `TwoWire.begin()` method takes two **int** arguments where in your implementation two **uint8_t** arguments are provided which results in incorrect I2C pin configuration and wrong sensor measurement values reported on ESP32.
Specifying **int8_t** arguments works as the underlying HAL initialisation method takes those.

See the [`begin()` method in Wire.cpp](https://github.com/espressif/arduino-esp32/blob/a9d77ac66e7190333433ed4c90966aeaac0a0d6c/libraries/Wire/src/Wire.cpp#L181).
See also the [`i2cInit` method in esp32-hal-i2c.h](https://github.com/espressif/arduino-esp32/blob/bd83727cd7dc3452fab2a3d83503dd96eca1ad4b/cores/esp32/esp32-hal-i2c.h#L28).

Suggested solution:
The constructor should be changed from
`bool begin(const uint8_t dataPin, const uint8_t clockPin)`
to
`bool begin(const int8_t dataPin, const int8_t clockPin)`

RobTillaart, 2022-06-21:
As the  ESP32 1.0.6 package defines
```cpp
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
```

I will fix it to 

```cpp
bool begin(const int dataPin, const int clockPin)
```

RobTillaart, 2022-06-21:
Develop branch created with fix. 
+ a fix for the **getEIDB()**

RobTillaart, 2022-06-21:
@jannvck 
Do you have time to confirm the develop branch is working for you?

jannvck, 2022-06-23:
Thank you for the quick fix!
I can confirm the develop branch is working for me. The sensor reports proper values now.
I have not tested getEIDB(), though.

closed 2022-06-23 by jannvck

RobTillaart, 2022-06-23:
Thanks for verifying, I will merge a.s.a.p

commit 408ebb02 (2022-06-23): fix #9 ESP32 Wire.begin + fix getEIDB() (#10)