RobTillaart/SHT2x #16: Create ASYNC interface for the SHT2X library

RobTillaart, 2022-12-14:
#### Problem

The read is blocking for quite a long time, depending on the resolution
In mutitasking and or real time environments this is not acceptable.

#### Request

An asynchronous interface like the SHT85 library has with the following functions:
-  bool requestData();
-  bool dataReady();
-  bool readData();

RobTillaart, 2022-12-14:
@countrysideboy 

I'll try to write a first version tonight, as I have no hardware can you test it for me and provide feedback?

There will be separate async calls for temperature and humidity

- ** bool requestTemperature()**
- ** bool requestHumidity()**
- ** bool reqTempReady()**
- ** bool reqHumReady()**
- ** bool readTemperature()**
- ** bool readHumidity()**

countrysideboy, 2022-12-14:
I am glad to test it, but I am not at home these days.

RobTillaart, 2022-12-14:
> I am glad to test it, but I am not at home these days.

you may test is anywhere 😎 as I have no hardware to test.

Created a branch with my first trial code - https://github.com/RobTillaart/SHT2x/tree/async_test
Build is running.
I added one minimal example that just does temperature. 

There is at  least one thing that need a fix and that is that the library is not aware of the type of request it is waiting for.
So for now the user should keep track of which type of request is pending. 

I'll try to dive into the datasheet later this week to see if the sensor can do a bit better. I have some raw notes about the async interface but never took time to elaborate them. 

(as it is rather late here I'll be back tomorrow)

RobTillaart, 2022-12-14:
Oops build failure, fix that first

countrysideboy, 2022-12-16:
I use the code from SHT2x_demo_async.ino, but the temperature always return -46.85 degree.(si7021)
I will try to compile SHT2x_demo_async.ino and flash it to a dev board, and report the result in recent days.

BTW, how to read temperature and humidity with the asynchronous interface?

countrysideboy, 2022-12-16:
`#include <Wire.h>
#define I2C_ADDR 0x40

// I2C commands
#define RH_READ             0xE5
#define RH_READ_NOHOLD      0xF5
#define TEMP_READ           0xE3
#define TEMP_READ_NOHOLD    0xF3
#define POST_RH_TEMP_READ   0xE0

#define RESET               0xFE
#define WRITE_USER_REG_1    0xE6
#define READ_USER_REG_1     0xE7

#define HTRE        0x02
#define _BV(bit) (1 << (bit))

/* ======================================================================
  Function: si7021_checkCRC
  Purpose : check the CRC of received data
  Input   : value read from sensor
  Output  : CRC read from sensor
  Comments: 0 if okay
  ====================================================================== */
uint8_t si7021_checkCRC(uint16_t data, uint8_t check)
{
  uint32_t remainder, divisor;

  //Pad with 8 bits because we have to add in the check value
  remainder = (uint32_t)data << 8;

  // From: http://www.nongnu.org/avr-libc/user-manual/group__util__crc.html
  // POLYNOMIAL = 0x0131 = x^8 + x^5 + x^4 + 1 : http://en.wikipedia.org/wiki/Computation_of_cyclic_redundancy_checks
  // 0x988000 is the 0x0131 polynomial shifted to farthest left of three bytes
  divisor = (uint32_t) 0x988000;

  // Add the check value
  remainder |= check;

  // Operate on only 16 positions of max 24.
  // The remaining 8 are our remainder and should be zero when we're done.
  for (uint8_t i = 0 ; i < 16 ; i++) {
    //Check if there is a one in the left position
    if ( remainder & (uint32_t)1 << (23 - i) )
      remainder ^= divisor;

    //Rotate the divisor max 16 times so that we have 8 bits left of a remainder
    divisor >>= 1;
  }
  return ((uint8_t) remainder);
}

void si7021_writeReg(uint8_t value)
{
  // Write to user register on ADDRESS
  Wire.beginTransmission(I2C_ADDR);
  Wire.write(WRITE_USER_REG_1);
  Wire.write(value);
  Wire.endTransmission();
}

uint8_t si7021_readReg()
{
  // Read from user register on ADDRESS
  Wire.beginTransmission(I2C_ADDR);
  Wire.write(READ_USER_REG_1);
  Wire.endTransmission();
  Wire.requestFrom(I2C_ADDR, 1);
  uint8_t regVal = Wire.read();
  return regVal;
}

void si7021_reset()
{
  //Reset user resister
  si7021_writeReg(RESET);
}

void si7021_resolution(uint8_t i)
{
  // Changes to resolution of ADDRESS measurements.
  // Set i to:
  //      RH         Temp
  // 0: 12 bit       14 bit (default)
  // 1:  8 bit       12 bit
  // 2: 10 bit       13 bit
  // 3: 11 bit       11 bit

  uint8_t regVal = si7021_readReg();
  // zero resolution bits
  regVal &= 0b011111110;
  switch (i) {
    case 1:
      regVal |= 0b00000001;
      break;
    case 2:
      regVal |= 0b10000000;
      break;
    case 3:
      regVal |= 0b10000001;
    default:
      regVal |= 0b00000000;
      break;
  }
  // write new resolution settings to the register
  si7021_writeReg(regVal);
}

void si7021_heaterOn()
{
  // Turns on the ADDRESS heater
  uint8_t regVal = si7021_readReg();
  regVal |= _BV(HTRE);
  //turn on the heater
  si7021_writeReg(regVal);
}

void si7021_heaterOff()
{
  // Turns off the ADDRESS heater
  uint8_t regVal = si7021_readReg();
  regVal &= ~_BV(HTRE);
  si7021_writeReg(regVal);
}

bool si7021_begin(const int dataPin, const int clockPin)
{
  Wire.begin(dataPin, clockPin);
  Wire.beginTransmission(I2C_ADDR);
  //reset
  Wire.write(WRITE_USER_REG_1);
  Wire.write(RESET);
  if (Wire.endTransmission() == 0) {
    return true;
  }
  return false;
}

void si7021_request_humi() {
  Wire.beginTransmission(I2C_ADDR);
  Wire.write(RH_READ_NOHOLD);
  Wire.endTransmission(false);
}

float si7021_get_humi() {
  Wire.requestFrom(I2C_ADDR, 3);
  uint16_t hum = Wire.read();
  hum <<= 8;
  hum |= Wire.read();
  uint8_t chxsum = Wire.read();

  // Check CRC of data received
  if (si7021_checkCRC(hum, chxsum) != 0) {
    return -100;
  }

  float humidity = hum;
  humidity *= 125;
  humidity /= 65536;
  humidity -= 6;

  // fix humidity
  if (humidity > 100) humidity = 100.0f;
  else if (humidity < 0) humidity = 0.1f;

  return humidity;
}

// Read temperature from previous RH measurement.
void si7021_request_temp_prev() {
  Wire.beginTransmission(I2C_ADDR);
  Wire.write(POST_RH_TEMP_READ);
  Wire.endTransmission(false);
}

float si7021_get_temp_prev() {
  Wire.requestFrom(I2C_ADDR, 2);
  uint16_t temp = Wire.read();
  temp <<= 8;
  temp |= Wire.read();

  float temperature = temp;
  temperature *= 175.72;
  temperature /= 65536;
  temperature -= 46.85;

  return temperature;
}

void si7021_request_temp() {
  Wire.beginTransmission(I2C_ADDR);
  Wire.write(TEMP_READ_NOHOLD);
  Wire.endTransmission(false);
}

float si7021_get_temp() {
  Wire.requestFrom(I2C_ADDR, 3);
  uint16_t temp = Wire.read();
  temp <<= 8;
  temp |= Wire.read();
  uint8_t chxsum = Wire.read();

  // Check CRC of data received
  if (si7021_checkCRC(temp, chxsum) != 0) {
    return -100;
  }

  float temperature = temp;
  temperature *= 175.72;
  temperature /= 65536;
  temperature -= 46.85;

  return temperature;
}`

RobTillaart, 2022-12-16:
If you use the default demo sketch, does that work?

It is rather late so I will try to understand the 46.85 result. 

What is the purpose of the copy paste code you posted?

countrysideboy, 2022-12-17:
I can confirm that the code works for my si7021 now.
I use the code below:
sht2x.requestTemperature();
PT_DELAY(85); // a macro provide from protothreads.h
sht2x.readTemperature();
sht2x.getTemperature();

As I can not find my dev board, I did not try the default demo sketch(SHT2x_demo_async.ino).
But I think the reason getting -46.85 degree is because of SHT2x_demo_async.ino do not perform the readTemperature() function, makes the _rawTemperature = 0(default value).

RobTillaart, 2022-12-17:
> I can confirm that the code works for my si7021 now.

Good to hear!

> But I think the reason getting -46.85 degree is because of SHT2x_demo_async.ino do not perform the readTemperature() function, makes the _rawTemperature = 0(default value).

I will have a look at the code, and fix it - writing code with no hardware has always risks like this (sorry)

After the fix I will merge the async branch into master

RobTillaart, 2022-12-17:
By any chance, did you verified the humidity part too?

countrysideboy, 2022-12-17:
> By any chance, did you verify the humidity part too?

Yes, the humidity part is also ok.

RobTillaart, 2022-12-17:
just pushed code with fix in example
Also added a field requestType to track if there is a TEMP or HUM request pending.
That will break the interface in the future as I think I want to have:

- **bool requestReady()** checks if a request is pending and handles the waiting time for the right type.
- **bool requestPending()** checks if a request is pending. 
- **uint8_t getRequestType()** returns 0, 1, 2

In the current code the requestType is reset in **readTemperature()** so one can check reqTempReady() multiple times.

Also thinking of 
- **int requestReady()** => -1 is no request pending, 0 is waiting, 1 is ready.

What is your opinion on this?

closed 2022-12-18 by RobTillaart

countrysideboy, 2022-12-18:
> just pushed code with fix in example Also added a field requestType to track if there is a TEMP or HUM request pending. That will break the interface in the future as I think I want to have:
>
> * [...]

The si7021 has a hold mode, it seems something relate to the scl gpio(when the temperature is measuring, the pin keep high or low output).But I think the no hold mode(delay some ms according to the datasheet) is already working well, requestPending is a good idea to deal with the multiple requests.

RobTillaart, 2022-12-18:
Thanks for the feedback!
Holding the SCL line sounds like it may block the I2C bus for other devices. (todo check datasheet)
So that is definitely not wanted in a RT/MT environment.

I will add the requestPending() and getRequestType() in the next release.