RobTillaart/SHT2x #18: setResolution doesn't call SHT2x_WRITE_USER_REGISTER

madl3x, 2023-03-25:
I haven't checked the SHTx datasheet, but in the function `setResolution`, when calling `writeCmd`, the command used is `SHT2x_READ_USER_REGISTER`:

https://github.com/RobTillaart/SHT2x/blob/71a1d9433f4595f74732c355119cdfbdc8bd314e/SHT2x.cpp#L480

Shouldn't it be `SHT2x_WRITE_USER_REGISTER`?

Also, this `SHT2x_WRITE_USER_REGISTER` compiler macro is defined but not used anywhere in the code.
A bit weird.

RobTillaart, 2023-03-26:
Thanks for the issue. I will dive into it asap, might take some days.

RobTillaart, 2023-03-26:
There is a (long open) issue with the resolution #13 
It could be related. As I have not investigated that yet l cannot say for sure. Finally  I do not have such sensor available for testing  (no excuse, but it does not help either).
That said, you might have pinpointed the cause.

RobTillaart, 2023-03-26:
Had a look and indeed the **SHT2x_WRITE_USER_REGISTER** should be used in both **setResolution()** and some heater functions.

I will create a develop branch + PR asap. 
Are you able to test that branch?

RobTillaart, 2023-03-26:
Created the develop branch + PR https://github.com/RobTillaart/SHT2x/pull/19

madl3x, 2023-03-26:
Thank you for your fast response!

> Finally I do not have such sensor available for testing
You should test all your library updates with the proper hardware.
Doing blind updates is never a good idea for open-source, especially in communities like Arduino where people are using libraries to avoid understanding too much of the underlying hardware.

RobTillaart, 2023-03-26:
> Thank you for your fast response!

You're welcome!

> You should test all your library updates with the proper hardware.
> Doing blind updates is never a good idea for open-source, especially in communities like Arduino ...

I know, 
you are 100 % right, also not a good idea in closed source, never, 

However, I have about 200 libraries written (partly on request) and for most of them I have bought the hardware to test but not for all (for various reasons). For this library I don't have the hardware, so in the readme.md file is a disclaimer. Same is true for all different boards, there are hundreds of them with sometimes minor differences, and I only have a few to test. 
(not an excuse, just reality)

closed 2023-03-29 by madl3x

RobTillaart, 2023-03-31:
Otherwise do not hesitate to ask questions, especially with a well done analysis/diagnosis - appreciated 👍