RobTillaart/SHT2x #20: cant get humidity and temperature at same time in async mode

kakhi-kotchauri, 2023-09-09:
for some reasons i cant get humidity and temperature at same time in async mode, is it even possible to get both temperature and humidity at same time in async mode ?

RobTillaart, 2023-09-10:
Don't know if it is possible for the sensor type you used.
Have you studied the datasheet of the sensor used?
What does is state with respect to your question?

This library cannot do it at the same time.

kakhi-kotchauri, 2023-09-10:
thank you for your reply, sensor i used is HTU21D,  adafruit library that i used previously is blocking type and adding around 100ms delay that is causing problems in my code,  your library in other hand is not blocking in async mode but if i try to get humidity and temp at same time nothing is returned.

 i have ide but i don't know if it will work, i think i can request temp first and store it, then i can add async delay using mills and request humidity data after i store both humidity and temp i can use them in my code and cycle will repeat, sensor reading speed is not a problem in my case but code blocking is

RobTillaart, 2023-09-10:
Your idea will work, just change the mindset a bit.
Think of temperature as last retrieved temperature and humidity as last retrieved humidity.

Then you should see the retrieval as a async process that constantly updates the last retrieved T and H.
It may make very well sense to have different update frequencies for T and H, however that depends on the needs of your project. 

You could use an int state = 0 to indicate that both T and H are "old". Then measure T and state become 1, the measure H and state becomes 2. 
State 2 indicates both measurements are "fresh". 
By using a timestamp of the measurements, the state will be reset to 0 after X minutes / seconds.

RobTillaart, 2023-09-10:
Created an issue to elaborate an async example.
(Later this week probably)

closed 2023-09-10 by kakhi-kotchauri

RobTillaart, 2023-09-10:
@kakhi-kotchauri 
Created a develop branch with new example and  (a bit) improved async interface

kakhi-kotchauri, 2023-09-10:
I already make something like your last async example and it is working perfectly. thank you again for maintaining this library, i think it is only one which supports async mode