RobTillaart/SHT2x #25: Compilation error: SHT21.h: No such file or directory

sakinacodes, 2023-11-18:
FQBN: arduino:avr:uno
Using board 'uno' from platform in folder: C:\Users\Sakina\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\Sakina\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...
C:\Users\Sakina\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Sakina\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\Sakina\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\standard C:\Users\Sakina\AppData\Local\Temp\arduino\sketches\52EAEC77C9D80ADDDA1ECA096D64DEB4\sketch\Automatic_Irrigation_System_Final_Code.ino.cpp -o nul
Alternatives for SHT21.h: []
ResolveLibrary(SHT21.h)
  -> candidates: []
D:\Masters\Third semester\IOT AND BIG DATA\Project\Second project\Automatic_Irrigation_SMS-main\Arduino\Automatic_Irrigation_System_Final_Code\Automatic_Irrigation_System_Final_Code.ino:2:10: fatal error: SHT21.h: No such file or directory
 #include <SHT21.h>
          ^~~~~~~~~
compilation terminated.

exit status 1

Compilation error: SHT21.h: No such file or directory

RobTillaart, 2023-11-18:
Thanks for reporting an issue.
I assume your question is how to solve this.

It looks like you did not install the library properly.
Please install it with the library manager, and try to compile one of the examples.

RobTillaart, 2023-11-24:
@sakinacodes 

Is the issue solved now?

RobTillaart, 2023-11-27:
As there is no answer I assume it is solved ==> close the issue.

closed 2023-11-27 by RobTillaart

sakinacodes, 2023-11-28:
Sorry for the late reply. Yeah, it is solved now.