RobTillaart/SHT2x #31: GY-21 Device gives odd results!!

plackettsj1, 2025-04-05:
Hiya, Just wanted to let you know I tried the library with a GY-21. I used the GY21 class, and whilst the device was found both the temperature and humidity were negative :-) 
I didn't have time to check anything other than  getTemperature and getHumidity. Apologies I didn't have the time for better diagnosis.
Please throw this out if its not helpful :-)

Cheers
Simon

RobTillaart, 2025-04-05:
> I need to compare the datasheets of the GY21 and the SHT21 to see if these are compatible.

Quick look shows commands are identical

plackettsj1, 2025-04-06:
> Can you post the code you used? (text no image)
> And its output (text, no image)

Hi,
I was in the process of getting you the code and I spotted *MY* error - I
did not do the read - I just got the values -

*My Sketch:*
#include "Wire.h"
#include "SHT2x.h"
GY21 gy21;

void setup()
{
  Serial.begin(115200);
  Wire.begin();
  gy21.begin();
}

void loop()
{
  delay(1000);
  Serial.print(gy21.getTemperature(), 1);
  Serial.print("\t");
  Serial.println(gy21.getHumidity(), 1);
  delay(1000);
}

*... and the missing line:*
*gy21.read();*

Cheers
Simon

RobTillaart, 2025-04-06:
Good to hear it is solved
You may close the issue